convertersystems/opc-ua-client #251: Read node value

techvoipit, 2023-06-28:
I would like to read a value variable with this code but I can't isolate it:
var readRequest = new ReadRequest
                {
                    // set the NodesToRead to an array of ReadValueIds.
                    NodesToRead = new[] {
                    // construct a ReadValueId from a NodeId and AttributeId.
                    new ReadValueId {
                        // you can parse the nodeId from a string.
                        NodeId = NodeId.Parse("ns=2;s=Target01.PLC.Memory_Bits.AGITATORE_FERMO"),
                        //NodeId = NodeId.Parse(VariableIds.Server_ServerStatus),
                        // variable class nodes have a Value attribute.
                        AttributeId = AttributeIds.Value
                    }
                }
                };
                // send the ReadRequest to the server.
                var readResult = await channel.ReadAsync(readRequest);
foreach (var result in readResult.Results)
                    {
                    Console.WriteLine(result);
                    var val = result.GetValueOrDefault<DataValue>();
                            Console.WriteLine("Value: " + val.Value);

                            // Timestamp del valore letto
                            Console.WriteLine("Timestamp: " + val.SourceTimestamp);

                            // Stato del valore letto
                            Console.WriteLine("Status: " + val.StatusCode);
                       
                    }

return True; status: 0x00000000; ts: 28/06/2023 16:43:19

How I can get only true ?
attached the node structure
[image]

dbonfilt, 2023-06-29:
Is this not working for you?

```csharp
var val = result.GetValueOrDefault<bool>();
```

In fact, in 3.1.1 at least, the GetValueOrDefault requires a type parameter (?).

techvoipit, 2023-06-29:
Thank you very much is working!!
Can you share me where I can find some documentation about which type  I can use for GetValueOrDefault

ismdiego, 2023-07-03:
You are welcome, glad to help.

The GetValueOrDefault<T> accepts any NET type. I mean, it will try to cast the read value to the NET type you expect.

I think you are thinking about the OPC-UA types, that you can see here: [OPC UA Built-in Data Types](https://reference.opcfoundation.org/Core/Part6/v104/docs/5.1.2)

Also, you can find a typical mapping from these data types to C# .NET counterparts here (warning: this is NOT official documentation from this library, take this only as an example): [Quick OPC Data Types in OPC-UA](https://opclabs.doc-that.com/files/onlinedocs/QuickOpc/Latest/User%27s%20Guide%20and%20Reference-QuickOPC/Data%20Types%20in%20OPC-UA.html)

closed 2023-11-26 by awcullen