convertersystems/opc-ua-client #260: Service that implements Workstation.UaClient with subscription crashes when network outages occur

jasonfarner, 2023-10-11:
We have a service that is using the subscription implementation of the Workstation.UaClient for two separate plcs at once. We have noticed if we block/unblock the network connection repeatedly (using a powershell script and firewall rules) the service will crash with the exception below.  I have attached the script we are using to test with as well.  Thanks.

CoreCLR Version: 4.700.22.55902
.NET Core Version: 3.1.32
Description: The process was terminated due to an unhandled exception.
Exception Info: System.AggregateException: One or more errors occurred. (Timeout occurred while processing the request.)
 ---> Workstation.ServiceModel.Ua.ServiceResultException: Timeout occurred while processing the request.
   at Workstation.ServiceModel.Ua.Channels.ClientSecureChannel.RequestAsync(IServiceRequest request, CancellationToken token)
   at Workstation.ServiceModel.Ua.SubscriptionServiceSet.PublishAsync(IRequestChannel channel, PublishRequest request, CancellationToken token)
   at Workstation.ServiceModel.Ua.Channels.ClientSessionChannel.PublishAsync(CancellationToken token)
   --- End of inner exception stack trace ---
   at System.Reactive.PlatformServices.ExceptionServicesImpl.Rethrow(Exception exception) in /_/Rx.NET/Source/src/System.Reactive/Internal/ExceptionServicesImpl.cs:line 19
   at System.Reactive.ExceptionHelpers.Throw(Exception exception) in /_/Rx.NET/Source/src/System.Reactive/Internal/ExceptionServices.cs:line 16
   at System.Reactive.Stubs.<>c.<.cctor>b__2_1(Exception ex) in /_/Rx.NET/Source/src/System.Reactive/Internal/Stubs.cs:line 16
   at System.Reactive.AnonymousSafeObserver`1.OnError(Exception error) in /_/Rx.NET/Source/src/System.Reactive/AnonymousSafeObserver.cs:line 62
   at System.Reactive.Sink`1.ForwardOnError(Exception error) in /_/Rx.NET/Source/src/System.Reactive/Internal/Sink.cs:line 61
   at System.Reactive.Sink`2.OnError(Exception error) in /_/Rx.NET/Source/src/System.Reactive/Internal/Sink.cs:line 94
   at System.Threading.Tasks.Dataflow.DataflowBlock.SourceObservable`1.ObserversState.NotifyObserversOfCompletion(Exception targetException)
--- End of stack trace from previous location where exception was thrown ---
   at System.Threading.Tasks.Dataflow.Internal.Common.<>c.<ThrowAsync>b__20_0(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()

[Firewall Flicker Script.txt](https://github.com/convertersystems/opc-ua-client/files/12868786/Firewall.Flicker.Script.txt)

jasonfarner, 2023-10-11:
It seems to be coming from this code here, just not sure how to handle it on our end as try/catch does not work. 

[image]

jasonfarner, 2023-10-13:
I believe this may be because we have more than on plc connection for each instance of our service (in this case 2 connections)  I have run a test by blocking the connection to the one of the plcs on startup of the service so that never connects.  Once the second plc connects I start running the above script to simulate a flickering connection.  I ran this test for over an hour and our service did not crash leading me to believe that it is because we have more than one plc connection per instance of the service.

awcullen, 2023-11-26:
In the last line of code below, I register a OnPublishError method when calling Subscribe.
This method should handle the exception you are seeing. 

```
   void onPublish(PublishResponse pr)
    {
        // loop thru all the data change notifications and log them.
        var dcns = pr.NotificationMessage.NotificationData.OfType<DataChangeNotification>();
        foreach (var dcn in dcns)
        {
            foreach (var min in dcn.MonitoredItems)
            {
                logger.LogInformation($"sub: {pr.SubscriptionId}; handle: {min.ClientHandle}; value: {min.Value}");
            }
        }
    }

    void onPublishError(Exception ex)
    {
        logger.LogInformation("Exception in publish response handler: {0}", ex.GetBaseException().Message);
    }

    var token = channel
        .Where(pr => pr.SubscriptionId == id)
        .Subscribe(onPublish, onPublishError);

jasonfarner, 2023-11-29:
Thanks!  I will give it a try and let you know how it goes.

closed 2023-12-10 by awcullen