convertersystems/opc-ua-client #262: Empty Value from a node from Siemens PLC

ShiddhamSharma, 2023-11-17:
Hi,
I am trying to read and write a value from a tag in my PLC.
Now I have tried opcua in python and am able to read and write the node but when I try opc-ua-client i am receivng empty values.
I don't understand the issue but it might be because I am giving wrong Node ID.
The C# code:
```
var channel = new ClientSessionChannel(
                clientDescription,
                null, // no x509 certificates
                new AnonymousIdentity(), // no user identity
                "opc.tcp://192.168.0.1:4840", // the public endpoint of a server at opcua.rocks.
                SecurityPolicyUris.None); // no encryption
                
            await channel.OpenAsync();
            // build a ReadRequest. See 'OPC UA Spec Part 4' paragraph 5.10.2
            var readRequest = new ReadRequest {
                // set the NodesToRead to an array of ReadValueIds.
                NodesToRead = new[] {
                    // construct a ReadValueId from a NodeId and AttributeId.
                    new ReadValueId {
                        // you can parse the nodeId from a string.
                        // e.g. NodeId.Parse("ns=2;s=Demo.Static.Scalar.Double")
                        NodeId = NodeId.Parse("ns=3;s=Alarms.Data[2].Message"),
                        // variable class nodes have a Value attribute.
                        AttributeId = AttributeIds.Value
                    }
                }
            };
            try
            {
                // send the ReadRequest to the server.
                var readResult = await channel.ReadAsync(readRequest);
                string val = readResult.Results[0].GetValueOrDefault<string>();
                MessageBox.Show(val, "Error", MessageBoxButton.OK, MessageBoxImage.Error);
            }
            catch (Exception ex)
            {
                MessageBox.Show(ex.Message, "Error", MessageBoxButton.OK, MessageBoxImage.Error);
            }
```
My Python Code:
`node = client.get_node('''ns=3;s="Alarms"."Data"[1]."Message"''')`

ShiddhamSharma, 2023-11-17:
So Parsing it as:
`NodeId.Parse("ns=3;s=\"Alarms\".\"Data\"[2].\"Message\"")`
solved it

closed 2023-11-17 by ShiddhamSharma

awcullen, 2023-11-24:
Great!  I am glad it’s working.