convertersystems/opc-ua-client #269: Nodes being dropped on Subcription

mozerlou, 2024-02-14:
Hello OPC UA community,

I hope you're doing well. I am currently facing an issue with my OPC UA client where, after a while, some nodes in my subscription are not being reinstated. It seems to be consistently happening to specific nodes, and I'm seeking assistance in troubleshooting and resolving this issue.

Details:

OPC UA Client: Workstation.UaClient 3.11
Server: TBA (getting more information)
Observations:

I have implemented a subscription mechanism in my code, and it generally works well.
Nodes are intermittently not being reinstated in the subscription after a certain period.
The issue seems to affect specific nodes consistently.
I have a publish request running on a timer every hour, however I have noticed that the subscription gets recreated and the subscription name changes (Original name = OriginalName2 = OriginalName3) overtimes with a number being added to it. I have my suspicion that this might be happening during a reconnect. 
I have noticed that our application does do a lot of reconnect to the OPCUA server (m_reconnectHandler.BeginReconnect) 

Any help would really be appreciated

Questions:

How does the OPC UA server handle the subscription lifecycle, and are there any specific settings or conditions that might affect the subscription over time?
How does the OPC UA server handle session reconnects? I have implemented a reconnect mechanism, but I want to ensure it is working as expected. 
Are there any server-side logs or diagnostics that might indicate why certain nodes are not being monitored?

awcullen, 2024-02-14:
If you have a publish request running on a timer every _hour_, I would not use the subscription service.  Instead, every hour, I would create a new client, call Read with the ReadValueList, and close the client.   I wouldn't keep connections open for an hour, if I wasn't using it.

mozerlou, 2024-02-14:
Currently I am using the publish request to keep the subscription alive through the night when theres very low activity on the machine.
As I understand subscription have a keep alive time and for this application, its really important to be able to get those notifications. 

Would you think our session reconnect might be causing the subscription to drop nodes? 
I have notice this only happens during periods of high activity on the machine/OPC UA server we are connected to.

awcullen, 2024-02-17:
Okay, well I would experiment by calling CreateSubscription with different values for:
requestedPublishingInterval
requestedLifetimeCount
requestedMaxKeepAliveCount
and learning which settings are more reliable.

This client does not handle the subscription lifecycle.   When you create a new Session, do you currently call TransferSubscriptions?

mozerlou, 2024-02-23:
Hey AwCullen, thank you for your swift answer, I didnt see your message until now sorry about that. 

So I have a little bit more logs which I hope will be helpful. 

With my subscription I am subscribed to HandleStateChanged- and i was able to see that during my last session reconnect my subscription was reinstated successfully. 
I was able to see that on HandleStateChange on session reconnect all of my nodes were being added back using SubscriptionChangeMask.ItemsAdded  .

However 30 minutes later, one of the nodes had stopped receiving notifications on it No reconnect had happened since. Which leads me to believe that it might just be a lost notification. 

Here are the settings you were talking about. I had experimented with them for a couple of weeks to see how long my subscription would last. 
Right now like I said earlier my subscription itself can last all night, works for days on end which is perfect for what I need. 

   Opc.Ua.Client.MonitoredItem monitoredItem = new Opc.Ua.Client.MonitoredItem(m_subscription.DefaultItem)
                {
                    DisplayName = item.Name,
                    StartNodeId = item.Node,
                    AttributeId = Attributes.Value,
                    MonitoringMode = MonitoringMode.Reporting,
                    SamplingInterval = 200,
                    QueueSize = 1,
                    DiscardOldest = true
                };
                
 m_subscription = new Subscription(m_session.DefaultSubscription)
            {
                DisplayName = subscriptionName,
                PublishingEnabled = true,
                PublishingInterval = 50,
                KeepAliveCount = uint.MaxValue,//I 
                LifetimeCount = uint.MaxValue,
                MaxNotificationsPerPublish = 100,
                Priority = 100
            };
            
       I have about 15 nodes on a subscription so its really not that intense. I have 3 subscriptions. 
       
       
       I really have no idea why its dropping right now I am looking at your documentation and trying to understand the Republish(uint 	retransmitSequenceNumber)	 function

awcullen, 2024-02-28:
The code you posted above uses the library project at https://github.com/OPCFoundation/UA-.NETStandard.  I'm glad you found some information there.  I am sure the authors can help you with this issue.

closed 2024-03-07 by awcullen