convertersystems/opc-ua-client #270: Could not connect to HTTPS endpoint

ondrejpazintelegrafia, 2024-02-14:
Hello.
I could not manage to connect to HTTPS endpoint using this library. 
There is allways exception: 'Cannot access a disposed object. Object name: 'socket'.'

The exception is thrown even when using discovery service to find endpoints.

This is my code:
var getEndpointsRequest = new GetEndpointsRequest
{   
    EndpointUrl = "https://myPC:62641/IntegrationObjects/ServerSimulator/",
    ProfileUris = new[] { TransportProfileUris.HttpsXmlOrBinaryTransport },
};
var getEndpointsResponse = await DiscoveryService.GetEndpointsAsync(getEndpointsRequest);

awcullen, 2024-02-14:
This package does not support HTTPS endpoints,  only OPC.TCP.  Sorry for confusion.

ondrejpazintelegrafia, 2024-02-15:
Do you plan to support HTTPS endpoints in the future?

awcullen, 2024-02-17:
I don't think so.  The OPC UA TCP binding with the binary encoding is fast and more widely available in industrial devices.

closed 2024-02-28 by awcullen