convertersystems/opc-ua-client #275: Problem on channel.OpenAsync() when using IP address

hclf99, 2024-04-09:
I see a strange behavior on connection to my server: When using the IP address e.g. opc.tcp://xxx.xxx.xxx.xxx:4840 sometimes I get connected sometimes not (randomly), but instead when I use the PC Network name e.g. opc.tcp://MYDEFPC:4840 it always connects without problems.
Using:
            channel = new ClientSessionChannel
            (
                clientDescription,
                // no x509 certificates
                null,
                // no user identity
                new AnonymousIdentity(),
                // the OPC server
                server,  ===> using addressee as described above
                // no encryption
                SecurityPolicyUris.None
            );
Using version 3.2.3 on NET 4.8

By the way: The older version of my application used "channel = new UaTcpSessionChannel" without problems for both addressing methods

awcullen, 2024-05-01:
Thanks for the bug report.  Could you share the exception you get when the client can't connect?

hclf99, 2024-07-15:
Unfortunately there is no exception occurring, the connection attempt simple times out.... for example if I use this:
               Task t = channel.OpenAsync();                          
                t.Wait(10000);
                ...

after 10 sec it still says: CommunicationState.Opening    and it stays forever in the opening state 

I tried different OPC-UA servers for connection tests, on some it works without any problem on some (e.g Siemens S7) not (timeout)